dynawo/dyn-grid-compliance-verification #2: Filtering method in the signal processing / Exclusion windows

louisg-rte, 2024-11-08:
As we already discussed during previous meetings, the signal processing chain of the tool includes a low-pass filter. As a consequence the curves are smoothed and the electrical variables start varying even before an event (fault appearance or clearing). This is a problem for the exclusion windows which should start only at the event (forgetting the recent short-term solution that has been implemented)
We tried to investigate some solutions and one of them could be to use a filter that, at a given time, is only based on the past signal and is not centred around this given time. (so this filter would have non zero phase delay) Here is what it looks:
[image]

The solid red line is our current filter (using filtfilt from scipy) while the dashed line is obtained with lfilter from Scipy.
We can see that the results are quite different and with this other filter the curves only start varying after the event, so the "transient phase" would coincide better with the exclusion window.

louisg-rte, 2024-11-08:
Response from JL: 

_"Thanks for the discussion. My point of view is that the tool should offer several filtering options to play with (most importantly, allow tweaking f_c), but at the same time provide some default option to be adopted as "the RTE standard sigproc for compliance testing" (for which RTE may decide how strictly it wants to follow IEC 61400-27-2, but that's another question).

I reviewed IEC 61400-27-2 and it is prescribes the type of filter (digital 2nd order critically damped), since it says Annex E is normative. But it doesn't say anything about double passes (which are typical to avoid time shift), or, more interestingly in my opinion, about piece-wise filtering. This is what I'm thinking: since the sharp transitions are going to be excluded anyway, why include them in the filtering process? Applying the filter piece-wise in each window separately will avoid the problem of this smoothing extending over we don't want it. After all, the rationale for filtering is to eliminate high-frequency fluctuations, noise, etc., over the "smooth" parts of the simulation (and the more I think about this, I think that applying filtering for abrupt transitions is not a good idea, because of the way that DAE solvers simulate these)."_

jl-marin, 2025-01-29:
Solved in PR #17.  The most significant change is that the filtering is **now performed separately in each window**:
* 3 windows in case of Fault Ride-Through tests: Pre-fault, During-fault, Post-fault
* 2 windows in case of step-response / reference-tracking tests: Pre-event, Post-event. 

The filtering is still performed with filtfilt (which processes the signal in both the forward and reverse directions) because it is zero-phase; using lfilter introduces phase delays and would mess with the timings of events.

We now also use Gustaffson's method (as Matlab does) to minimize the "transient" artifacts produced by filters at both extremes of the signal, and even provide a configurable exclusion window to make sure those possible filtering artifacts do not appear when comparing signals for the calculation of KPIs.

We also provide a choice of several filters, just to experiment how they may affect the resulting values of KPIs: bilinear 2nd-order critically-damped, Bessel, Butterworth, and Chebyshev.

closed 2025-01-29 by jl-marin