dynawo/dyn-grid-compliance-verification #3: Definition of step in PCS description for reference tracking tests

louisg-rte, 2024-11-08:
In the PCS description the reference tracking tests are defined as follow:
[image]
The parameter setpoint_step_value (line 66) corresponds to the step size (final value - initial value) but the initial value is not configurable and is defined (I guess) as the initial voltage at PDR which is here pdr_U (line 49).
We should add a parameter that corresponds to the initial value of the step function. The reason is that, depending on the activated control mode, we don't necessary have U = Uref (even in the steady state before the event). Indeed a typical control mode for WT is Uref = U + lambda Q (lambda being a fixed parameter). (<=>Mwpqmode = 3 in the IEC WPP model). If we decide to define Q ≠ 0 at the initialization, we'll have U≠Uref, so the intial value of Uref should stay configurable.

marcosmc, 2025-02-11:
We have implemented this change:

if we encounter a case of **Mwpqmode = 3** in the IEC WPP model and **PDR Q ≠ 0**, in the context of a voltage setpoint step change test, then we internally calculate the setpoint Uref as **Uref = U + lambda Q**, where lambda is _Kwpqu_ and U is the PDR U.

Please test once we merge this to the main branch and let us know.

louisg-rte, 2025-02-13:
Related to this issue we should consider to redefine the variable MagnitudeControlledByAVRPu for all the Dynawo models with a plant controller.
I suggest to define it as U+lambda.Q for all of these models (U being the voltage magnitude at PDR, Q being the reactive power given by the producer and lambda being an internal parameter of the producer model). Considering that this variable is a combination of several variables, we need to discuss if we need to change the Dynawo models so that this variable is computed in the producer model or is the calculation done internally in the tool.

marcosmc, 2025-02-27:
Done

louisg-rte, 2025-02-27:
I noticed that in src/dgcv/dynawo/dictionary/Power_Park.ini we both have MagnitudeControlledByAVRPu and VoltageDrop defined. Could you tell me what variable is finally shown in the report ? Is it MagnitudeControlledByAVRPu or "U + VoltageDrop * Q" ?

marcosmc, 2025-02-27:
In WECC Plant models the MagnitudeControlledByAVRPu is defined as the following example:
MagnitudeControlledByAVRPu = WTG4A_wecc_repc_QVCtrlErr_y

VoltageDrop is used in the tool to define the lambda in the equation: Uref = U + lambda * Q

We are going to reopen the issue because the IEC case remains to be resolved.

louisg-rte, 2025-02-27:
OK so lambda (WTG4A_Kc) is used only for the calculation of the setpoint value to be connected to AVRSetpointPu. And then the variable shown in the report is WTG4A_wecc_repc_QVCtrlErr_y, is that it ?

marcosmc, 2025-02-27:
Yes, it is just as you say.

louisg-rte, 2025-02-27:
For the IEC models the problem is that there is no existing variables corresponding to "U+lambda.Q". It could be created. Otherwise it could be internally computed by the tool.

marcosmc, 2025-02-27:
Ok, the idea is to calculate it in the tool while it is not available in the Dynawo model.

closed 2025-03-03 by marcosmc