dynawo/dyn-grid-compliance-verification #14: Change of variable names in IEC models

louisg-rte, 2024-12-11:


louisg-rte, 2024-12-11:
Some variables have changed name in the IEC models. I wanted to adapt this in Power_Park.ini and this brings me some questions:
- What is the base power (for pu) that is expected by this dictionnary ? In the modelica models some variables are in pu based on SNom (the nominal power of the unit) and others are based on SnRef (=100 MVA which is a common base power for all Modelica models). Which of them have to be written in this dictionnary ?
- For IEC WT models there is a variable called InjectedActivePower and another one called ActivePowerPu. Why do we need these 2 variables ? (same for reactive power)
[image]

- For IEC WPP models why is there no variable corresponding to active/reactive power ?
[image]

marcosmc, 2024-12-19:
The only conversions ​​calculated by the tool are the values needed ​​to initialize the models used (because we calculate a network power-flow); for the rest of the values, the value entered by the user is respected, so it is interpreted that it has been entered using the base that Dynawo expects for the model in each case.
We have verified that the dictionary contains variables that are no longer used, in PR #15 all variables that are no longer in use have been removed.

closed 2024-12-19 by marcosmc