dynawo/dyn-grid-compliance-verification #18: Windows installer

louisg-rte, 2025-01-10:
* The link to the windows installer (.exe file) that is in the readme is broken
* After downloading it, this file is considered as dangerous by the window firewall:
[image]
At the moment, I have not been able to install the tool on windows because I don't have the administrator rights on my Windows my machine. But I will try to ask for a permission to the IT department to do so.

marcosmc, 2025-01-15:
Fixed the links to the latest version in the README document.

jl-marin, 2025-01-15:
Apart from the broken link in the README, we investigated the issue. Here's the upshot:

- Avoiding this type of warning needs a **code signing certificate**. This link explains many details about it:  https://stackoverflow.com/questions/48946680/how-to-avoid-the-windows-defender-smartscreen-prevented-an-unrecognized-app-fro. Does RTE have a certificate for signing Windows executables?
- The bigger issue is that this is an **installer**, and therefore it needs Admin rights (there's no way around this: a non-privileged windows user cannot install applications in the OS). And once you're an admin, there's no problem in unblocking the executable:

[image]

So maybe we should add this to the windows installation instructions.

marcosmc, 2025-02-12:
Updated the README.md and manuals in PR #40.

closed 2025-02-12 by marcosmc