dynawo/dyn-grid-compliance-verification #19: Use a variable impedance for the voltage dip and voltage surge test cases

Philibert92, 2025-01-10:
Right now, the tool is using a variable voltage infinite bus to create controlled voltage surge and voltage dip in the related test cases.
In the DTR, we ask producers to use a variable impedance to do it following this picture :
[image]

We recently added a variable impedance in the dynawo library to be able to reproduce this simulation. Would it be possible to be able to use this configuration in the tool too ? (For performance verification as well as for model validation)

jl-marin, 2025-01-15:
We recall that we already discussed this issue some time ago:
- For **electrical performance verification**, Carmen was of the opinion that Fiches I6 & I7 were better implemented via a controllable InfiniteBus because then the voltage swell / dip at the PDR bus would be *exactly* as prescribed by those two Fiches.
- However, for **model validation** (Fiche I16) we asked Carmen again because producers would use their own tools and they would probably be modeling the grid differently. This are the Notes we keep from that session:

```
Discussion points from Tech Meeting 2022-02-17
==============================================

We discussed some questions we had on how to specify the model for the grid
side, for the cases involving faults or grid voltage sags/surges.

In the performance fiches for PPM, tests I6 and I7 use a controllable source
(Inf Bus from Table), but here in this case the Producers will be using their
own tool to produce the Reference Curves and they will be modeling the grid
differently, and therefore there will be significant differences between their
reference curves and the Dynawo curves.

Carmen clarified:

  * yes, this is indeed a problem and therefore it is intended that in this
    FicheI1x we will NOT be using the Inf-bus-from-Table object, but the Zv
    impedance instead. Which is normally the way they model these cases in EMT
    tools (Carmen will provide the EMT Fiche to see it as an illustration)

  * It is still being discussed whether it makes sense at all to include tests
    like I6 & I7 in the context of Model Validation. When testing for
    performance, I6 & I7 are intended to check whether the facility does not
    trip; by contrast, I5 is different, as it is intended to check the
    performance of the injections, and it seems more relevant for Model
    Validation, compared to the other two tests. The question is still being
    studied by RTE.
```

Now, in order to implement the change in Fiche I16 examples, we see the following:
- For Zone-1, voltage step tests are +/- 10% voltage step tests; therefore this poses no big problem.
- But Zone-3 voltage dip/swell tests (i.e., I6/I7) are not just a step test; they have a prescription for a more complex voltage "curve" at the PDR. Implementing this via a variable impedance Zv would involve several (many) calculations of Zv: remember that Zv is calculated via bisection search, simulating different values of Zv until attaining the desired voltage at the PDR.

We think we need to mull over it a bit more before we give a definte answer.

jl-marin, 2025-02-21:
As discussed in our Tech Meeting today (Feb 21), we should indeed consider this improvement in our short-term roadmap:

   * For tests of type I6 / I7 (performance and Zone3 model validation), offer the user **the option** of implementing the test by either method: (a) controllable InfBus; (b) Configurable table of Zv values (as discussed in the previous comment).

For the voltage swell/dip tests of Zone-1, we still need more feedback from RTE.

closed 2025-06-16 by marcosmc