dynawo/dyn-grid-compliance-verification #26: Troubles trying to change OC in performance tests

Philibert92, 2025-01-31:
I am trying to change the OCs in the performance tests. 

Especially for the I2 test. In the PPM WECC, of PCS I2, the initial power is 0.75 p.u. but should be 1 p.u.
Thus I tried to modify the operating conditions via the config.ini file. I added the following lines:

[image]

But it doesn't change anything in the tests when I regenerate the report. Is there a new compilation needed ? Do I need to relaunch the tool ? Or to define the OCs elsewhere ?

Thank you for your help.
Regards,
Philibert

marcosmc, 2025-02-03:
The configuration used is not correct. Can you indicate where this example was obtained from? We will correct it to avoid further confusion.

In the config.ini_ADVANCED file you can find an example of the expected syntax::
```dosini
## [PCS_RTE-I2.USetPointStep.AReactance.Model]
## # Reactance of the line connected to the PDR point
## line_XPu = a
## # SCR stands for Short Circuit Ratio
## #SCR =
## # PDR point
## pdr_P = Pmax
## pdr_Q = 0
## pdr_U = Udim
## # Infinite Bus configuration
## # To configure time parameters, the following convention is used:
## # * 'delta_t_': indicates how long the network remains in a certain state, this value will be
## #                added to the time in which the event is triggered.
## # * otherwise: the value of this variable will be used in the tool without prior treatments.
## # TSO Model configuration
```

The configuration to be used is:
```dosini
[PCS_RTE-I2.USetPointStep.AReactance.Model]
pdr_P = 0.5*Pmax
```

Philibert92, 2025-02-06:
Thank you for the correction. I don't remember where I obtained the configuration I used but I should have used the config.ini_ADVANCED file as you said.

closed 2025-02-06 by Philibert92