dynawo/dyn-grid-compliance-verification #27: Run Windows version

louisg-rte, 2025-01-31:
I've succesfully installed the tool on Windows thanks to the DGCV windows installer. I'm using a completely empty virtual machine that I created for this purpose. All the third-parties (Dynawo, Python, Miktex, CMake ...) have been installed without any problems. Nevertheless when I run the DGCV file that has been created on the Desktop, it opens a terminal with the python environment loking activated but the "dgcv" command is not recognized.

[image]

marcosmc, 2025-02-03:
Added the following note to the system requirements section for installing the tool on a Windows system:

**NOTE:** The windows installer (described in the next section) will install all of these system requirements, so you may skip this section. It is only here for your information.

marcosmc, 2025-02-03:
Can you check if the **c:\DGCV\dgcv_venv\Scripts** directory is similar to the following image?

[image]

Is it possible for you to send us the zipped dgcv_venv directory?

louisg-rte, 2025-02-03:
[image]

[dgcv_venv.zip](https://github.com/user-attachments/files/18646520/dgcv_venv.zip)

jl-marin, 2025-02-04:
With respect to:

> One small suggestion regarding this would be to write clearly in the readme that these third-parties don't have to be manually installed by the user and that everything will be done by the installer itself !

I changed the order of the two Windows subsections and added more noticeable notes about this.  Please check if the explanation is better now.

marcosmc, 2025-02-12:
The Windows installer has been modified for version 0.7.0.

Please try the new release and let us know.

louisg-rte, 2025-02-24:
I succeed to install the tool but I'm still unable to use it. It seems that the issue is related to Dynawo. The Dynawo models can't be compiled:

[image]

It may be related to a Dynawo issue that you reported weeks ago ? https://github.com/dynawo/dynawo/issues/3502

marcosmc, 2025-02-24:
Indeed, your user in Windows has a space, and Dynawo is not prepared to work with paths that contain a space. You can check this yourself by going into any of the directories created by Windows in the path:
`C:\Users\Louis Garbay\AppData\Local\dgcv\ddb`

louisg-rte, 2025-02-24:
[image]

marcosmc, 2025-02-24:
Can you send us the log files that were generated when compiling the models? These are located in the same path where they are compiled

louisg-rte, 2025-02-28:
it seems to be linked to a space in the path once again