dynawo/dyn-grid-compliance-verification #31: Performance tests failing in the examples

Philibert92, 2025-02-06:
Several simulations are failing in the examples.

The ones I found were the following (but there might be others):

dgcv/examples/PPM/Dynawo/IEC2015 : I10 is failing

[image]

dgcv/examples/PPM/Dynawo/IEC2020 : I5 and I10 are failing

[image]

dgcv/examples/PPM/Dynawo/WECC : I7 and I10 are failing
[image]

This might be a problem with the models, the solvers or with dynawo more than with the dgcv tool itself. However, I still think we should work for those examples tests to be fully working.

marcosmc, 2025-02-20:
Tests I5 and I7 have been fixed.

marcosmc, 2026-04-15:
This issue is currently considered dependent on the Dynawo integration (PR #3952).

The underlying initialization problems are expected to be resolved once Dynawo is integrated into `master`, as the initialization sequence will change accordingly.

However, this issue will remain open until the Dynawo integration (PR #3952) is available and the behavior is validated.