dynawo/dyn-grid-compliance-verification #36: Definition of BusPDR_BUS_ActivePower and BusPDR_BUS_ReactivePower

louisg-rte, 2025-02-11:
From what I investigated BusPDR_BUS_ActivePower and BusPDR_BUS_ReactivePower are internally defined in the tool as the P and Q at the line terminal. Could you confirm ?
The problem is that in the line Dynawo model the base power is SnRef (=100 MVA) but I think it is more consistent to consider the base power of the unit. So I suggest to get these two variables from the producer model instead of the line model. (and make sure that the ones from the producer model have SNom as base power)
What do you think ?

louisg-rte, 2025-02-19:
I have thought about this again and I realized that (P,Q) at the PDR are not necessary (P,Q) at the output of the producer model (we can have transformer and "internal network" between PDR and output of producer model). So it may be better to keep using (P,Q) from the line terminal but apply the base power change ( * SnRef / SNom) so that the curves shown in the report are based on the producer base power

marcosmc, 2025-02-27:
The **BusPDR_BUS_ActivePower** and **BusPDR_BUS_ReactivePower** curves are still obtained from the equipment connected directly to the **PDR bus** on the producer model side, before being displayed in the final report the following base change is performed:  

> BusPDR_BUS_ActivePower = active_power * SnRef / SNom
> BusPDR_BUS_ReactivePower = reactive_power * SnRef / SNom
>
> where:
> 1. **BusPDR_BUS_ActivePower** and **BusPDR_BUS_ReactivePower** are the tool curves
> 2. **active_power** and **reactive_power** are the curves obtained from the equipment connected to the PDR Bus (Dynawo) 
>

closed 2025-02-27 by marcosmc