dynawo/dyn-grid-compliance-verification #42: Generate function sending error

Philibert92, 2025-02-12:
I tried using the generate function. I filled the DYD file the following way : 

[image]

But then when I press enter in the terminal I get the following error : 

[image]

I work with a developer installation of the tool (with the cloned repository) and I use an up to date version of dynawo.
This error didn't appear until recently but I think you modified the function in the last version of the tool and there might be a new bug.

marcosmc, 2025-02-13:
The tool is designed to provide the user with a **PAR** file that already contains all the parameters that must be filled in for each model used in the **DYD** file, including the parameters with default values, although the latter already have a value assigned (the same default value). These parameters that the **PAR** file must contain are extracted by the tool directly from _Dynawo_, specifically from the **.XML.DESC** files present in the simulator's model library. 

Up to the latest version of the simulator that we have tested, these files are always present, but for some reason in your installation this does not seem to be the case. Without these files (**.XML.DESC**), the **generate** command is no longer useful to the user, since the **PAR** file would be an empty template.

Philibert92, 2025-02-13:
Yes, I understand. However, the XML.DESC are actually present in my installation but the generate command doesn't find them and thus creates an empty PAR file. Did you try it on your machine ?
If you cannot reproduce this error, I can create a branch for you to reproduce it.

marcosmc, 2025-02-14:
We have just verified that when compiling dynawo, the **ddb** directory is located in a different path than the **myEnvDynawo.sh** command that you use.
In our case this directory is in the path: `dynawo/install/gcc11/master/Release/shared/dynawo`
Two possible solutions:

- Run the tool with the full path to the dynawo.sh command that is generated when compiling Dynawo.
`dgcv generate -o modele_single -t S -v performance_PPM -l /home/philinux/dynawo/dynawo/install/gcc11/master/Release/shared/dynawo/dynawo.sh`
- Add the full path to the dynawo build to the PATH variable
`export PATH=/home/philinux/dynawo/dynawo/install/gcc11/master/Release/shared/dynawo:$PATH`