dynawo/dyn-grid-compliance-verification #45: WECC_WTGB I2 test case working on dynawo but not with the tool

Philibert92, 2025-02-14:
[WECCWTG4BCurrentSource_I2_ProducerPar.zip](https://github.com/user-attachments/files/18800122/WECCWTG4BCurrentSource_I2_ProducerPar.zip)

Hello. In this zip file, I put dynawo files reproducing the I2 test case. It runs well with the appropriate solver parameters.

I tried to run the same case with the tool with the following files.

[WECCB_producer_parameters.zip](https://github.com/user-attachments/files/18800594/WECCB_producer_parameters.zip)

However I have 2 problems:

1. The simulations fail easily, even when I tried to modify the _dgcv_repo/src/dgcv/model_lib/solvers.par_ accordingly to the parameters I used in my dynawo simulation (I played with "minimalAcceptableStep" and "accuracy" mainly)
2. If the simulation doesn't fail, the report doesn't show the variable I wanted in the report (I modified the Power_park.ini to use _MagnitudeControlledByAVRPu = WTG4B_wecc_repc_QVCtrlErr_y_ for the _WTG4BWeccCurrentSource_ model). This variable is the true U+Lambda.Q that should be compared to URef in the simulation. But I cannot show it in the report and I don't understand why.

I tried to share my dgcv branch with you by pushing it on git but I didn't have the rights. I hope the zip I put here and the explanations are enough.

marcosmc, 2025-02-20:
We have checked the example of the issue with the main branch of the tool and the result obtained is the following:

[image]

The main branch already contains the implementation of the backup mode, initially the case failed so in a first step the tool tries to simulate the case again decreasing the **minStep** and **minimalAcceptableStep**, despite the changes the simulation fails. so the tool modifies the **absAccuracy** and **relAccuracy** parameters, finally with these changes the simulation is carried out without problems.

```bash
dgcv performance -m Dynawo -o results2 -p PCS_RTE-I2
Do you want to overwrite the results directory? (y/[N]]) y
2025-02-20 08:42:06,606 |                DGCV.Validation |       INFO |             validation.py:   89 | DGCV Electric Performance Verification for Power Park Modules
2025-02-20 08:42:06,624 |                 DGCV.Benchmark |       INFO |              benchmark.py:  545 | RUNNING BENCHMARK: PCS_RTE-I2.USetPointStep, OPER. COND.: AReactance
2025-02-20 08:42:06,880 |                    DGCV.Dynawo |    WARNING |                 curves.py:  553 | launching job 'PCS I2 (Power Park Modules) - Voltage Setpoint Step'
DYN Error: IDA fails to solve the equations ( DYNSolverIDA.cpp:500 )
Dynawo execution failed

2025-02-20 08:42:06,881 |                    DGCV.Dynawo |    WARNING |                 curves.py:  554 | Retry by modifying the minimum time step
2025-02-20 08:42:07,117 |                    DGCV.Dynawo |    WARNING |                 curves.py:  592 | launching job 'PCS I2 (Power Park Modules) - Voltage Setpoint Step'
DYN Error: IDA fails to solve the equations ( DYNSolverIDA.cpp:500 )
Dynawo execution failed

2025-02-20 08:42:07,118 |                    DGCV.Dynawo |    WARNING |                 curves.py:  593 | Retry by modifying the required accuracy
2025-02-20 08:42:07,889 |                 DGCV.Benchmark |       INFO |              benchmark.py:  545 | RUNNING BENCHMARK: PCS_RTE-I2.USetPointStep, OPER. COND.: BReactance
2025-02-20 08:42:10,684 |                    DGCV.Report |       INFO |                 report.py:  356 | 
Summary Report
==============

***Run on 2025-02-20 08:42 CET***
***Dynawo version: 1.8.0 (rev:master-ae4d969)***
***Model: Dynawo***

Pcs          Benchmark                Operating Condition      Overall Result
-----------------------------------------------------------------------------
PCS_RTE-I2   USetPointStep            AReactance               Compliant
PCS_RTE-I2   USetPointStep            BReactance               Compliant

2025-02-20 08:42:12,630 |                  DGCV.PDFLatex |       INFO |                 report.py:  485 | PDF done.
2025-02-20 08:42:12,654 |                DGCV.Validation |       INFO |             validation.py:   45 | Opening the report: results2/Reports/report.pdf
```

Regarding the variable **MagnitudeControlledByAVRPu** , we have modified the master dictionary as indicated in the issue and the result obtained is as expected:

[image]

So we have analyzed the Dynawo log files included in the attached case. When opening the file we noticed the following warning provided by Dynawo:

```log
2025-02-14 15:28:57 | INFO | -----------------------------------------------------------------------
2025-02-14 15:28:57 | INFO | starting curve initialization
2025-02-14 15:28:57 | INFO | -----------------------------------------------------------------------
2025-02-14 15:28:57 | WARN | curve not added: Id: Wind_Turbine , name: WTG4B_wecc_repc_QVCtrlErr_y #It should be WTG4B_measurements_UPu + WTG4B_Kc*WTG4B_measurements_QPu
2025-02-14 15:28:57 | INFO | end of curve initialization, added 10 curves
2025-02-14 15:28:57 | INFO | -----------------------------------------------------------------------
```

The error is that INI files do not allow comments to be used on the same line as a variable is defined, this comment has caused the curve file for Dynawo to be incorrect:
```xml
<?xml version='1.0' encoding='UTF-8'?>
<curvesInput xmlns="http://www.rte-france.com/dynawo">
  <curve model="BusPDR" variable="bus_terminal_V_re"/>
  <curve model="BusPDR" variable="bus_terminal_V_im"/>
  <curve model="StepUp_Xfmr" variable="transformer_terminal1_V_re"/>
  <curve model="StepUp_Xfmr" variable="transformer_terminal1_V_im"/>
  <curve model="StepUp_Xfmr" variable="transformer_terminal1_i_re"/>
  <curve model="StepUp_Xfmr" variable="transformer_terminal1_i_im"/>
  <curve model="Wind_Turbine" variable="WTG4B_injector_idPu"/>
  <curve model="Wind_Turbine" variable="WTG4B_injector_iqPu"/>
  <curve model="Wind_Turbine" variable="WTG4B_wecc_repc_URefPu"/>
  <curve model="Wind_Turbine" variable="WTG4B_wecc_repc_QVCtrlErr_y #It should be WTG4B_measurements_UPu + WTG4B_Kc*WTG4B_measurements_QPu"/>
  <curve model="Wind_Turbine" variable="WTG4B_omegaRefPu"/>
</curvesInput>
```

We will modify the tool to protect it from similar cases.

marcosmc, 2025-02-20:
Done in PR #48

closed 2025-02-20 by marcosmc