dynawo/dyn-grid-compliance-verification #64: Xa and Xb not affected when changing SNom

louisg-rte, 2025-03-10:
It seems that the values of Xa and Xb on TSO side are not affected when changing the value of SNom. Xa and Xb are defined in the DTR in pu in base UNom , SNom. So when the nominal power of the producer SNom is modified, it should change the value of Xa and Xb. (dividing SNom by 10 will muliply Xa and Xb by 10 because Xnom = Unom^2 / SNom)
This could be (one of) the reason that are causing problems when trying to change the value of SNom in our examples.

marcosmc, 2025-03-10:
In the code, _Xa_ and _Xb_ are determined by the following formula: **x_dtr * (u_dim**2 / u_nom**2) * (s_nref / s_nom)**
where:
- x_dtr is the line reactance defined in the DTR.
- udim is the generator sizing voltage (Udim) by its nominal voltage.
- u_nom is the nominal voltage in the PDR Bus. (Producer.ini)
- s_nref is the System-wide S base (SnRef).
- s_nom is the nominal apparent power of all generating units. (Producer.ini)

Any change in SNom should modify the final value of the reactance, we will review it.

louisg-rte, 2025-03-10:
I guess that the problem is that I was changing SNom only in the Producer.par (internal parameter of Dynawo model) but not the SNom in the Producer.ini

louisg-rte, 2025-03-11:
I close this issue and open a new one more appropriate

closed 2025-03-11 by louisg-rte