dynawo/dyn-grid-compliance-verification #75: Test tagged wrongly as non compliant

Philibert92, 2025-03-18:
There are case where the KPIs seem valid (here in red 0.0514 < 0.08) but there are tagged non compliant. (test Transient fault:High-impedance fault (V = 0.5U n, 800ms))

[image]

Also there are cases where the KPIs seem wrongly calculated : (test Permanent fault: High-impedance fault (SCR = 10, V = 0.5U n))

[image]

[image]

Could you investigate ? (I sent you the test case by email)

Philibert92, 2025-04-11:
Another example of KPIs wrongly calculated for a zone 3 test : 

[image]

[image]

[image]

marcosmc, 2025-06-18:
Solve in PR #104