dynawo/dyn-grid-compliance-verification #80: Question - Modify the setpoint value for reference tracking tests

Philibert92, 2025-04-07:
As we have seen in issue #63, in some models, the electrical quantity controlled might not be at the model terminal.
This leads to wrong initialization for reference tracking tests because the "step_value0" obtained for the setPoint in the TSOModel.par comes from the initial value at the model terminal (P0, Q0 or U0).

I wonder if we could define the model variable refering to the electrical quantity controlled for each reference tracking test. This way, we would use the initial value of this variable for the "step_Value0"in the TSOModel.par.

Do you see a way to do this easily on the short term ? (it doesn't have to be user friendly at first, I would just need it to run some tests)

Regards,
Philibert

louisg-rte, 2025-04-09:
I would say that we could re-use the values that are computed in the internal _INIT file of the dynawo model, because this "delta" is already computed there. For example in PVCurrentSource_INIT:
`  u0Pu = ComplexMath.fromPolar(U0Pu, UPhase0);`
`  s0Pu = Complex(P0Pu, Q0Pu);`
`  i0Pu = ComplexMath.conj(s0Pu / u0Pu);`
`  iInj0Pu = - i0Pu * SystemBase.SnRef / SNom;`
`  uInj0Pu = u0Pu - Complex(RPu, XPu) * i0Pu;`
`  sInj0Pu = uInj0Pu * ComplexMath.conj(iInj0Pu);`
`  PInj0Pu = ComplexMath.real(sInj0Pu);`
`  QInj0Pu = ComplexMath.imag(sInj0Pu);`

here `(P0Pu,Q0Pu,U0Pu,UPhase0)` are defined at the output terminal of the model and `(PInj0Pu,QInj0Pu)` are defined behind the impedance `(RPu+jXPu)` and this is where the setpoints `(PInjRefPu,QInjRefPu)` are applied **for WECC models without plant control**. So if we get these values we could define step_value0 = PInj0Pu for P and step0_value = QInj0Pu for Q.

louisg-rte, 2025-04-09:
But I'm not sure this is feasible because these values (PInj0Pu,QInj0Pu) are computed only when the Dynawo model is running but here we need these values to build the test case so it's a kind of vicious circle.