dynawo/dyn-grid-compliance-verification #81: Prepare the tool to treat F16 sheet

Philibert92, 2025-04-07:
(Not urgent - longterm)

We need to think about how to handle the F16 sheet in dycov (comparing phasor signals from generic models to real electrical measurements).

Among other things, we will need to : 
- Work on measured electrical signals and see if they need treatment to be compared to generic model simulated signals. (RTE will provide for anonymized measured signals). 
- Define a new report/a new command for this sheet.

Regards,
Philibert

marcosmc, 2026-09-28:
Since Fiche F16 selects no test of its own, F16 is shipped as a declaration template rather than a catalogue: `PCS_RTE-F16z1` and `PCS_RTE-F16z3` declare no test and carry, commented out, a declaration with one operating condition per active power level of the fiche, which the producer completes with the tests actually recorded and delivers next to the reference curves of the case. A PCS that declares no test is left out of the run.

The `Wind/IECB2015` and `BESS/WECC` examples carry a declaration invented for the example, with records anonymized from the tool's own simulations, so that F16 runs end to end.

closed 2026-09-29 by marcosmc

marcosmc, 2026-10-08:
RTE accepted the proposal (2026-10-06) with three requests: the file delivered with the records is named `F16Description.ini`; F16 applies to Zone 3 only; and a tutorial plus manual chapters must explain every compliance test the tool has and how `F16Description.ini` activates each kind of test, so that a producer can write the declaration from the documentation alone.

Decisions taken to meet them: `PCS_RTE-F16z1` is removed; the Zone 1 model is demanded only when a PCS of zone 1 is validated, so a producer in the commissioning phase delivers `Zone3/` alone and runs `-p PCS_RTE-F16z3`; the template of the declaration ships as `F16Description.ini` next to the PCS description and is copied under that same name into the user's templates at start-up, ready to copy beside the reference curves and complete; and the voltage dip and swell tables ship with `PCS_RTE-F16z3`, so every Zone 3 kind of test (setpoint step, three-phase fault, voltage dip, voltage swell, frequency ramp, islanding) can be declared, the dip and swell benchmarks keeping the names of their tables.