dynawo/dyn-grid-compliance-verification #84: BESS performance and model validation - Missing operating points

Philibert92, 2025-04-15:
I ran DyCoV on a BESS performance test case using WECC BESS. The problem is that DyCoV generates only simulations with the operating point P=Pmax. However, the DTR states that with storage, you also need to simulate with P=-Pmax (when the battery is charging) for most performance sheets.

Philibert92, 2025-04-22:
You can reproduce this issue with example /home/philinux/dycov_repo/examples/Model/BESS/WECC/Dynawo/Zone3

trying the following commands :

(dycov_venv) philinux@philinux-virtual-machine:~/dycov_repo/examples/Model/BESS/WECC/Dynawo$ dycov performance -l /home/philinux/dynawo/dynawo/myEnvDynawo.sh -m Zone3 -o Results_perf/

or 

(dycov_venv) philinux@philinux-virtual-machine:~/dycov_repo/examples/Model/BESS/WECC$ dycov validate -l /home/philinux/dynawo/dynawo/myEnvDynawo.sh -m Dynawo -o Results ReferenceCurves

marcosmc, 2025-04-25:
As in the DTR two different values ​​are defined for the active power of the battery:
1. P = [Pmax injection]
2. P = [Pmax consumption]

It is proposed to modify the _Producer.ini_ file to allow the user to define both values.

marcosmc, 2025-04-30:
For **zone 1** of **PCS I16** of the DTR, it is not specified whether it is necessary to perform the tests with _Pmax_injection_unite_ and _Pmax_consumption_unite_ or only with one of them, the same happens for **PCS I10**. In all these cases, Pmax_unite is mentioned without specifying which Pmax it refers to (injection or consumption) or whether the tests must be performed with both values.

Philibert92, 2025-05-06:
You are right, the DTR is not clear in this respect. 
For both case (zone 1 of PCS I16 and PCS 10), DyCoV should perform the tests with _Pmax_injection_unite_ **and** _Pmax_consumption_unite_.