dynawo/dyn-grid-compliance-verification #116: Dynawo Modelica Library modification

louisg-rte, 2025-07-18:
### Suggestion

Following dynawo PR https://github.com/dynawo/dynawo/pull/3554 some modifications have been made to the Dynawo Modelica Library (Photovoltaics WECC models)

- PhotovoltaicsWeccVoltageSourceA is now PhotovoltaicsWeccVoltageSource1 (with internal control modifications in REGC-B)
- PhotovoltaicsWeccVoltageSourceB is now PhotovoltaicsWeccVoltageSource2 (with internal control modifications in REGC-B)
- PhotovoltaicsWeccCurrentSourceB is now PhotovoltaicsWeccCurrentSource (with internal control modifications in REGC-B)
- PhotovoltaicsWeccVoltageSource3 has been added (same as PhotovoltaicsWeccVoltageSource1 but with REGC-C instead of REGC-B)
- PhotovoltaicsWeccVoltageSource4 has been added (same as PhotovoltaicsWeccVoltageSource2 but with REGC-C instead of REGC-B)

- REGC-B has been modified to be a real voltage source interface. Previously, this was actually a current source interface.
- REGC-C is also a voltage source interface with the grid but this includes an additionnal control loop (which is called current loop) after the calculation of (Id_ref, Iq_ref)
- I don't think the external variables have been modified. But some parameters have probably changed or be added. You can find dynawo examples with these models in dynawo/examples/DynaSwing/WECC/PV
- Each of these models has it twin with "NoPlantControl" which is just the same model but without REPC. This corresponds in dycov to zone 1 vs zone 3.
- Initialization correction of the voltage regulation has been also made in this PR. URef0 is now correctly computed as U0 + lambda . Q0 with the correct signs

louisg-rte, 2025-07-18:
Additionnally, model WTG4AWeccCurrentSource2 has been added which very similar to WTG4AWeccCurrentSource1 (which was previously named WTG4AWeccCurrentSource) but with a small variant in the mechanical model of the turbine. There is no change in the external variables and there is only one additionnal parameter in WTG4AWeccCurrentSource2 which is WTG4A_tp (typical value 0.5)

marcosmc, 2026-02-17:
For additional details on how Zone 1 is implemented for WECC Type 3 models, please refer to issue #204, where the implementation approach and rationale are explained.